sci-bots/microdrop-3 #1: Device upload has no effect

cfobel, 2017-09-29:
# Setup

 1. Create new empty Conda environment:

        conda create -n microdrop-3
 2. Add `lucaszw` Conda channel:

        conda add --channels lucaszw
 3. Install `microdrop-3-bundle`:

        conda install microdrop-3-bundle
 4. Run `microdrop-3`.
 5. Launch `http://localhost:3000/plugin-manager` in browser.
 6. Enable the following plugins in the plugin manager:
     - `device_info_plugin` (process)
     - `droplet_planning_plugin` (process)
     - All UI plugin manager plugins
 7. Launch `http://localhost:3000/display` in browser.
 8. Upload the following device using the device upload widget **(unzip first)**: [device.zip](https://github.com/sci-bots/microdrop-3.0/files/1345348/device.zip)

# Expected behaviour

Device map is displayed in device UI panel.

# Observed behaviour

No change in display and no errors reported.

# Detailed env info

Conda packages:

```
# packages in environment at C:\Users\chris\Miniconda2\envs\microdrop-3:
#
certifi                   2017.7.27.1              py27_0    conda-forge
conda-helpers             0.12.2                   py27_0    sci-bots
decorator                 4.1.2                    py27_0    conda-forge
dmf-device                0.1.post4                     0    lucaszw
droplet-planning          0.2.post1                     0    wheeler-microfluidics
flatland-fork             0.4.post2                     0    wheeler-microfluidics
flatland-helpers          0.1.post2                py27_0    lucaszw
libiconv                  1.14                      vc9_4  [vc9]  conda-forge
libxml2                   2.9.5                     vc9_0  [vc9]  conda-forge
libxslt                   1.1.29                    vc9_5  [vc9]  conda-forge
lxml                      4.0.0                    py27_0    conda-forge
microdrop-3               0+unknown            ha837359_0    lucaszw
microdrop-3-bundle        0+unknown                     0    lucaszw
microdrop.default-ui-plugins 0.0.1.post2                   0    lucaszw
microdrop.device-info-plugin 0.1.post47                    0    lucaszw
microdrop.device-loader-ui-plugin 0+unknown                     0    lucaszw
microdrop.droplet-planning-plugin 2.1.post57                    0    lucaszw
microdrop.experiment-ui-plugin 0+unknown                     0    lucaszw
microdrop.protocol-execution-ui-plugin 0+unknown                     0    lucaszw
microdrop.protocol-ui-plugin 0+unknown                     0    lucaszw
mkl                       2017.0.3                      0
mqtt-messages-python      v0.0.1                   py27_0    lucaszw
networkx                  2.0                      py27_0    conda-forge
nodejs                    8.2.1                         0    sci-bots
ntfsutils                 0.1.3                         0    wheeler-microfluidics
numpy                     1.13.1                   py27_0
onoff                     1.0.1                    py27_0    lucaszw
paho-mqtt                 1.2.3                    py27_0    wheeler-microfluidics
paho-mqtt-helpers         0.2+5                    py27_0    lucaszw
pandas                    0.20.3                   py27_1    conda-forge
pandas-helpers            0.2                      py27_0    sci-bots
path_helpers              0.5.post1                py27_0    wheeler-microfluidics
pint                      0.7.2                         0    wheeler-microfluidics
pip                       9.0.1                    py27_0    conda-forge
pymunk                    4.0.0                         0    wheeler-microfluidics
pyparsing                 2.2.0                    py27_0    conda-forge
python                    2.7.13                        1    conda-forge
python-dateutil           2.6.1                    py27_0    conda-forge
pytz                      2017.2                   py27_0    conda-forge
pyyaml                    3.12                     py27_1    conda-forge
scipy                     0.19.1              np113py27_0
setuptools                36.3.0                   py27_0    conda-forge
si-prefix                 0.4.post10                    0    wheeler-microfluidics
six                       1.11.0                   py27_1    conda-forge
svg-model                 0.9.1                         0    wheeler-microfluidics
svgwrite                  1.1.8                         0    wheeler-microfluidics
vc                        9                             0    conda-forge
vs2008_runtime            9.00.30729.5054               0
vs2015_runtime            14.0.25420                    0    conda-forge
wheel                     0.30.0                     py_1    conda-forge
wheezy.routing            0.1.157                  py27_0    lucaszw
wincertstore              0.2                      py27_0    conda-forge
yaml                      0.1.6                     vc9_0  [vc9]  conda-forge
zlib                      1.2.11                    vc9_0  [vc9]  conda-forge
```

Conda environment:

```
Current conda install:

               platform : win-32
          conda version : 4.3.23
       conda is private : False
      conda-env version : 4.3.23
    conda-build version : 2.1.17
         python version : 2.7.13.final.0
       requests version : 2.14.2
       root environment : C:\Users\chris\Miniconda2  (writable)
    default environment : C:\Users\chris\Miniconda2\envs\microdrop-3
       envs directories : C:\Users\chris\Miniconda2\envs
                          C:\Users\chris\AppData\Local\conda\conda\envs
                          C:\Users\chris\.conda\envs
          package cache : C:\Users\chris\Miniconda2\pkgs
                          C:\Users\chris\AppData\Local\conda\conda\pkgs
           channel URLs : https://conda.anaconda.org/lucaszw/win-32
                          https://conda.anaconda.org/lucaszw/noarch
                          https://conda.anaconda.org/microdrop-plugins/win-32
                          https://conda.anaconda.org/microdrop-plugins/noarch
                          https://conda.anaconda.org/sci-bots/win-32
                          https://conda.anaconda.org/sci-bots/noarch
                          https://conda.anaconda.org/wheeler-microfluidics/win-32
                          https://conda.anaconda.org/wheeler-microfluidics/noarch
                          https://conda.anaconda.org/conda-forge/win-32
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.continuum.io/pkgs/free/win-32
                          https://repo.continuum.io/pkgs/free/noarch
                          https://repo.continuum.io/pkgs/r/win-32
                          https://repo.continuum.io/pkgs/r/noarch
                          https://repo.continuum.io/pkgs/pro/win-32
                          https://repo.continuum.io/pkgs/pro/noarch
                          https://repo.continuum.io/pkgs/msys2/win-32
                          https://repo.continuum.io/pkgs/msys2/noarch
            config file : C:\Users\chris\.condarc
             netrc file : None
           offline mode : False
             user-agent : conda/4.3.23 requests/2.14.2 CPython/2.7.13 Windows/10 Windows/10.0.15063
          administrator : False
```

Lucaszw, 2017-09-29:
Should be fixed now, seems to have been caused by the microdrop-3 npm package being outdated.

cfobel, 2017-10-02:
Confirmed that this is fixed by upgrading the `microdrop-3` conda package from `0+unknown-ha837359_0` to `v3.0.22-h08c1b4f_0`.

closed 2017-10-02 by cfobel