sci-bots/microdrop-3 #4: error(dev-setup): error setting up dev environment

cfobel, 2018-02-02:
While running the `npm run bootstrap` command in the **[Installing From Source (Latest)][1]** section of the [`microdrop-3` `README`][2], I encountered an error.

Here is the complete console output from the `npm run bootstrap` command: [2018.02.02-dev setup error.log](https://github.com/sci-bots/microdrop-3/files/1689690/2018.02.02-dev.setup.error.log)

Here is one of the log files referenced in the error output which may be relevant: [2018-02-02T14_24_12_171Z-debug.log](https://github.com/sci-bots/microdrop-3/files/1689694/2018-02-02T14_24_12_171Z-debug.log)

[1]: https://github.com/sci-bots/microdrop-3/blob/f1efd63f45201b7ebd165e66ca1d1c52a8af5ace/README.md#installing-from-source-latest
[2]: https://github.com/sci-bots/microdrop-3/blob/f1efd63f45201b7ebd165e66ca1d1c52a8af5ace/README.md

Lucaszw, 2018-02-02:
I updated the README instructions for building from source (I was missing a --recursive flag)

My suspicion is that lerna was installing an old version of micropede (since it was not cloned recursively), which still had an accidental node-gyp dependency. 

Try:
```
git clone --recursive https://github.com/sci-bots/microdrop-3
cd microdrop-3
npm i --global lerna
npm install
npm run upgrade:micropede
npm run bootstrap
npm run build
npm run start
```

cfobel, 2018-02-02:
I still get the same error when I use the steps from the [latest README][1] (from 7feff7aa1ff2dbfc03d21652dd1d68ad154a568e).

Here is the full error log from the console: [2018.02.02-dev setup error-7feff7aa.log](https://github.com/sci-bots/microdrop-3/files/1690083/2018.02.02-dev.setup.error-7feff7aa.log)

I also tried again after running the optional `npm run upgrade:micropede` command, but got the same error.  Here is the full error log from the console: [2018.02.02-dev setup error-7feff7aa-with upgraded micropede.log](https://github.com/sci-bots/microdrop-3/files/1690109/2018.02.02-dev.setup.error-7feff7aa-with.upgraded.micropede.log)

------------------------------

# Missing `ncu` requirement for `npm run upgrade:micropede`

Also, I noticed that to run `npm run upgrade:micropede`, I first needed to install `ncu`.  Otherwise I would receive this error:

```sh
(md-3-dev) PS C:\Users\chris\Documents\github\microdrop-3> npm run upgrade:micropede

> @ upgrade:micropede C:\Users\chris\Documents\github\microdrop-3
> lerna exec -- ncu -u @micropede/client @micropede/broker

lerna info version 2.8.0
'ncu' is not recognized as an internal or external command,
operable program or batch file.
lerna 'ncu' is not recognized as an internal or external command,
operable program or batch file.
ERR!'ncu' is not recognized as an internal or external command,
operable program or batch file.
 exec Errored while executing 'ncu -u @micropede/client @micropede/broker' in '@micropede/broker'
lerna ERR! execute callback with error
lerna ERR! Error: spawn ncu ENOENT
lerna ERR!     at notFoundError (C:\Users\chris\Miniconda2-x86\envs\md-3-dev\node_modules\lerna\node_modules\cross-spawn\lib\enoent.js:11:11)
lerna ERR!     at verifyENOENT (C:\Users\chris\Miniconda2-x86\envs\md-3-dev\node_modules\lerna\node_modules\cross-spawn\lib\enoent.js:46:16)
lerna ERR!     at ChildProcess.cp.emit (C:\Users\chris\Miniconda2-x86\envs\md-3-dev\node_modules\lerna\node_modules\cross-spawn\lib\enoent.js:33:19)
lerna WARN complete Waiting for 2 child processes to exit. CTRL-C to exit immediately.
lerna ERR! exec Errored while executing 'ncu -u @micropede/client @micropede/broker' in '@microdrop/buildutils'
lerna ERR! exec Errored while executing 'ncu -u @micropede/client @micropede/broker' in '@micropede/mixins'
{ Error: spawn ncu ENOENT
    at notFoundError (C:\Users\chris\Miniconda2-x86\envs\md-3-dev\node_modules\lerna\node_modules\cross-spawn\lib\enoent.js:11:11)
    at verifyENOENT (C:\Users\chris\Miniconda2-x86\envs\md-3-dev\node_modules\lerna\node_modules\cross-spawn\lib\enoent.js:46:16)
    at ChildProcess.cp.emit (C:\Users\chris\Miniconda2-x86\envs\md-3-dev\node_modules\lerna\node_modules\cross-spawn\lib\enoent.js:33:19)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
  errno: 'ENOENT',
  code: 'ENOENT',
  syscall: 'spawn ncu',
  killed: false,
  stdout: null,
  stderr: null,
  failed: true,
  signal: null,
  cmd: 'ncu -u @micropede/client @micropede/broker',
  timedOut: false,
  exitCode: 1 }
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @ upgrade:micropede: `lerna exec -- ncu -u @micropede/client @micropede/broker`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @ upgrade:micropede script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\chris\AppData\Roaming\npm-cache\_logs\2018-02-02T16_40_54_614Z-debug.log
```

The `README` should be updated to reflect this.

[1]: https://github.com/sci-bots/microdrop-3/blob/7feff7aa1ff2dbfc03d21652dd1d68ad154a568e/README.md#installing-from-source-latest

cfobel, 2018-02-02:
Same issue after cloning 8cf86a411f7018a50447b593e73e2f66c5c797c5.  Here's the console log output:
 [2018.02.02T14h52-dev setup error-8cf86a41-with upgraded micropede.log](https://github.com/sci-bots/microdrop-3/files/1690646/2018.02.02T14h52-dev.setup.error-8cf86a41-with.upgraded.micropede.log)

Might it have something to do with a specified version dependency for `node-sass`?  The snippet from the log below suggests that the file `win32-ia32-57_binding.node` from [`node-sass`](https://github.com/sass/node-sass/releases) from version **3.13.1** is required, but that file does not exist in the [`node-sass` **3.13.1** GitHub release][ns3.13.1].  However, the `win32-ia32-57_binding.node` _does exist_ in the [**4.5.3** `node-sass` release][ns4.5.3].

Any thoughts on how to resolve the `node-sass` issue?

---------------------------

Log snippet showing `node-sass` error:

```
....
lerna info version 2.8.0
lerna info Bootstrapping 11 packages
lerna info lifecycle preinstall
lerna info Installing external dependencies
lerna ERR! execute callback with error
lerna ERR! Error: Command failed: npm install
lerna ERR! Cannot download "https://github.com/sass/node-sass/releases/download/v3.13.1/win32-ia32-57_binding.node":
lerna ERR!
lerna ERR! HTTP error 404 Not Found
lerna ERR!
lerna ERR! Hint: If github.com is not accessible in your location
lerna ERR!       try setting a proxy via HTTP_PROXY, e.g.
lerna ERR!
lerna ERR!       export HTTP_PROXY=http://example.com:1234
lerna ERR!
lerna ERR! or configure npm proxy via
lerna ERR!
lerna ERR!       npm config set proxy http://example.com:8080
lerna ERR! gyp info it worked if it ends with ok
...
```

[ns3.13.1]: https://github.com/sass/node-sass/releases/tag/v3.13.1
[ns4.5.3]: https://github.com/sass/node-sass/releases/tag/v4.5.3

Lucaszw, 2018-02-02:
Should work now after updating dat.GUI . Also: I had a mistake in the instructions (should install npm-check-updates globally instead of ncu [seems to be a weather package ;p ])

cfobel, 2018-02-05:
Thanks @Lucaszw !  I am no longer getting any error following the README development set up instructions.

When I run `npm run start`, the following window is now launched:

[image]

**_However_**, when I try to browse to http://localhost:3000 my browser says the site can't be reached:

[image]

I'll look into this to see what I can figure out, but wanted to mention it here in case you've come across this before and have a quick fix.

Lucaszw, 2018-02-05:
There can sometimes be a delay between when the application loads that message, and when the web server is ready. If you reload the page ~10s after the message appears, does the web-page still not load?

Lucaszw, 2018-02-05:
I pushed a few changes to master that might help resolve the issue, can you try pulling and see if things still seem to be broken?

cfobel, 2018-02-05:
Thanks for your help.  I tried reloading several minutes after launching using `npm run start` and still no luck.

After updating to c0d6186e489eece943558673b1720f0de400161a, I still cannot load the page.  In addition, I now get an error about not finding `backbone`.  Note that I still get a node process running in a separate window:

[image]

Here is the full console log:

```sh
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3> npm run start

> @ start C:\Users\chris\Documents\GitHub\microdrop-3
> npm run --prefix ./packages/application start

> @microdrop/application@1.1.25 start C:\Users\chris\Documents\GitHub\microdrop-3\packages\application
> node index.js

Launching Electron APP
Launching Models
module.js:538
    throw err;
    ^

Error: Cannot find module 'backbone'
    at Function.Module._resolveFilename (module.js:536:15)
    at Function.Module._load (module.js:466:25)
    at Module.require (module.js:579:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (C:\Users\chris\Documents\GitHub\microdrop-3\Micropede\packages\client\src\client.js:4:18)
    at Module._compile (module.js:635:30)
    at Object.Module._extensions..js (module.js:646:10)
    at Module.load (module.js:554:32)
    at tryModuleLoad (module.js:497:12)
    at Function.Module._load (module.js:489:3)
module.js:538
    throw err;
    ^

Error: Cannot find module 'backbone'
    at Function.Module._resolveFilename (module.js:536:15)
    at Function.Module._load (module.js:466:25)
    at Module.require (module.js:579:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (C:\Users\chris\Documents\GitHub\microdrop-3\Micropede\packages\client\src\client.js:4:18)
    at Module._compile (module.js:635:30)
    at Object.Module._extensions..js (module.js:646:10)
    at Module.load (module.js:554:32)
    at tryModuleLoad (module.js:497:12)
    at Function.Module._load (module.js:489:3)
module.js:538
    throw err;
    ^

Error: Cannot find module 'backbone'
    at Function.Module._resolveFilename (module.js:536:15)
    at Function.Module._load (module.js:466:25)
    at Module.require (module.js:579:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (C:\Users\chris\Documents\GitHub\microdrop-3\Micropede\packages\client\src\client.js:4:18)
    at Module._compile (module.js:635:30)
    at Object.Module._extensions..js (module.js:646:10)
    at Module.load (module.js:554:32)
    at tryModuleLoad (module.js:497:12)
    at Function.Module._load (module.js:489:3)
```

Lucaszw, 2018-02-05:
It looks like I was missing backbone as a dependency in @micropede/client . That might have been responsible for the issue earlier. Could you try again after pulling the latest commit (make sure to also pull the latest from the Micropede submodule ({root}/Micropede) .

cfobel, 2018-02-05:
# Setup

 1. Display console log window by setting `show: true` in `BrowserWindow`
    constructor for `mainWindow` in `microdrop-3/packages/application/app.js`.
 2. Under `microdrop-3` root, run `npm run start`.

# Observed behaviour

Javascript console log displayed errors indicating the following packages were missing
from `Micropede\packages\broker`:

 - `ioredis`
 - `mongodb`
 - `msgpack-lite`

For example:

[image]

# Workaround

 1. Under `Micropede\packages\broker`, run: `npm install ioredis mongodb msgpack-lite`
 2. Under `microdrop-3` root, run `npm run start`.

# Fix (to do)

@Lucaszw is looking into a proper fix for this.

Lucaszw, 2018-02-05:
Appears to be due to old dependencies hoisted into the main repos and some redis and mongodb requirements in ascoltatori (fixed now via a pkg fork). Closing this issue.

closed 2018-02-05 by Lucaszw