sci-bots/microdrop-3 #5: test(application): verify `node` is not required to be installed for packaged electron app

cfobel, 2018-02-05:
In [`packages/application/index.js`][app], some child processes are [launched using `node <script name>`][node-launch-lines].

**It seems like this would require `nodejs` to be installed in the environment where the packaged Electron app is running.**

# Hypothesis

The bundled `microdrop-3` Electron app fails to load if `node` is not installed/available in the environment where the app is running.

# Test

 1. Bundle `microdrop-3` as an executable Electron app (e.g., using `electron-builder` or `electron-packager`).
 2. Ensure `node` is **NOT** on system path (e.g., rename `node.exe` to something else).  Running `cmd /C where node` at a command prompt should not show *any* output.
 3. Run `microdrop-3` executable.
 4. Verify `microdrop-3` app launches as expected.

[app]: https://github.com/sci-bots/microdrop-3/blob/47d2ebf9f85cbe20a1d42e2eca525be104ed0305/packages/application/index.js
[node-launch-lines]: https://github.com/sci-bots/microdrop-3/blob/47d2ebf9f85cbe20a1d42e2eca525be104ed0305/packages/application/index.js#L19-L21

Lucaszw, 2018-02-05:
I'm looking into updating this soon (the building process is still actively in development). There are allot of comments on how to handle forked child process' for built electron apps which I'm looking into.

Lucaszw, 2018-02-07:
Removed all spawned child processes in favour of electron windows.

closed 2018-02-07 by Lucaszw