sci-bots/microdrop-3 #6: docs(electron-packager): add instructions with steps to build exe using `electron-packager`

cfobel, 2018-02-05:
From the reading I have done, it seems like one of the strengths of basing an app around Electron is that the resulting app can be easily bundled in one of several ways, e.g., using [`electron-builder`][eb], [`electron-packager`][ep], etc.

This flexibility seems great; while tools like [`electron-builder`][eb] provide a way to generate a full installer with almost no configuration, others like [`electron-packager`][ep] generate a bundled executable application that can, e.g.,  be distributed as a portable application, included in a custom installer, etc.

Since both of these tools have their uses, it would be great for `microdrop-3` to be compatible either.

@Lucaszw , as far as I know, you have `microdrop-3` successfully building as an Electron executable app installer using [`electron-builder`][eb] (assuming all is well with #5 ).  **Therefore, this issue focuses on writing instructions for building an Electron executable app using [`electron-packager`][ep].**

---------------------------------------------------------

# Example

For the electron/electron-quick-start app, an Electron executable can be bundled as follows:

```sh
git clone https://github.com/electron/electron-quick-start
npm install -g electron-packager
npm install
electron-packager .
```

---------------------------------------------------------

# To do

Write instructions like in the **Example** section above to build a `microdrop-3` Electron executable app **using [`electron-packager`][ep]**.

[eb]: https://github.com/electron-userland/electron-builder
[ep]: https://github.com/electron-userland/electron-packager

cfobel, 2018-02-05:
# Notes

I am able to create an Electron executable using the following commands:

```sh
git clone --recursive https://github.com/sci-bots/microdrop-3
cd microdrop-3
npm i --global lerna npm-check-updates
npm install
npm run bootstrap
cd packages/application
electron-packager . microdrop-3
```

_However_, when I run execute the resulting `microdrop-3.exe` in a shell (from the generated `microdrop-3-win32-ia32` directory), things do not execute as I expected.

Observations:

 - Control immediately returns to the shell, but the following output is still printed to the console (also see the full console log output below):
    ```sh
    Launching Electron APP
    Launching Models
    ```
 -  It looks like two background processes are left running (listed using `Get-WmiObject Win32_Process | Select-Object ProcessID, CommandLine | ft -wrap -autosize` in Powershell):
    ```sh
    ...
     9624 "C:\Users\chris\Documents\GitHub\microdrop-3\packages\application\microdrop-3-win32-ia32\microdrop-3.exe"
     5092 "C:\Users\chris\Documents\GitHub\microdrop-3\packages\application\microdrop-3-win32-ia32\microdrop-3.exe" --type=gpu-process --no-sandbox --supports-dual-gpus=false --gpu-driver-bug-workarounds=7,10,19,23,41,61,74
          --disable-gl-extensions="GL_KHR_blend_equation_advanced GL_KHR_blend_equation_advanced_coherent" --gpu-vendor-id=0x8086 --gpu-device-id=0x5916 --gpu-driver-vendor="Intel Corporation" --gpu-driver-version=21.20.16.4664 --gpu-driver-date=4-21-2017
          --service-request-channel-token=3006C55D3DDE2F41A42C6A85C1E9B5D8 --mojo-platform-channel-handle=1712 /prefetch:2
    ...
    ```

----------------------------------------------------------------------------

# Console log

Here is the full console log (**note: the console is not blocked after the last command, i.e., the `microdrop-3` app is running in a background process**):

```sh
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3\packages\application> electron-packager . microdrop-3
Packaging app for platform win32 ia32 using electron v1.7.10
Wrote new app to C:\Users\chris\Documents\GitHub\microdrop-3\packages\application\microdrop-3-win32-ia32
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3\packages\application> cd .\microdrop-3-win32-ia32\
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3\packages\application\microdrop-3-win32-ia32> .\microdrop-3.exe
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3\packages\application\microdrop-3-win32-ia32>
Launching Electron APP
Launching Models
```

Lucaszw, 2018-02-05:
I'm working on fixing a few issues similar to this by separating the "launcher" component of microdrop/application into its own package (as most of the issues seem to revolve around handling resources wrt the root package). 

[@microdrop/microdrop](https://github.com/sci-bots/microdrop-3/tree/master/packages/microdrop)

Hopefully this will resolve some of the issues you are seeing with electron-packager as well.

Lucaszw, 2018-02-07:
I added an electron-packager config file in microdrop-builder (tested and works with lerna / symlinked packages)

cfobel, 2018-02-07:
@Lucaszw please add a README to the `microdrop-builder` directory documenting how to build the Electron app

Lucaszw, 2018-02-09:
https://github.com/sci-bots/microdrop-3/tree/master/microdrop-builder

closed 2018-02-09 by Lucaszw