sci-bots/microdrop-3 #12: feat(command-line): add command-line interface

cfobel, 2018-02-07:
Add command-line interface.

For example:

```
usage: MicroDrop.exe [-h] {launch,save,open,load_device} ...

positional arguments:
  {launch,save,open,load_device}
                        commands
    launch              Launch MicroDrop
    save                Save MicroDrop state to a file
    open                Open saved MicroDrop state file (launch MicroDrop if
                        not already running)
    load_device         Load device file

optional arguments:
  -h, --help            show this help message and exit

-----------------------------------------------------

usage: MicroDrop.exe launch [-h] [--host HOST]
                           [--http-port HTTP_PORT]
                           [--mqtt-tcp-port MQTT_TCP_PORT]
                           [--mqtt-ws-port MQTT_WS_PORT]

optional arguments:
  -h, --help            show this help message and exit
  --host HOST           MicroDrop application server host (default=localhost)
  --http-port HTTP_PORT
                        HTTP port (default=3000)
  --mqtt-tcp-port MQTT_TCP_PORT
                        MQTT TCP port (default=1884)
  --mqtt-ws-port MQTT_WS_PORT
                        MQTT websockets port (default=8083)

-----------------------------------------------------

usage: MicroDrop.exe save [-h] output_file

positional arguments:
  output_file  output file path

optional arguments:
  -h, --help   show this help message and exit

-----------------------------------------------------

usage: MicroDrop.exe open [-h] input_file

positional arguments:
  input_file  input file path

optional arguments:
  -h, --help  show this help message and exit

-----------------------------------------------------

usage: MicroDrop.exe load_device [-h] device_file

positional arguments:
  device_file  device file, .svg

optional arguments:
  -h, --help   show this help message and exit
```

Lucaszw, 2018-02-09:
Added with b5b67572133e72553df8e6c5b8de2b2a05b801f4

cfobel, 2018-02-12:
See [here][1] for how to implement command-line sub-commands using the `yargs` library.

[1]: https://github.com/yargs/yargs/blob/master/docs/advanced.md#commands

closed 2018-02-14 by Lucaszw