sci-bots/microdrop-3 #8: bug(packager): zombie processes left running after closing app

cfobel, 2018-02-07:
# Set up

 1. Build Electron executable:

     ```sh
     cd microdrop-builder
     npm run packager
     ```
 2. Run Electron executable
     ```sh
     cd Microdrop-win32-ia32
     Microdrop.exe
     ```
 3. Close application window.

# Expected behaviour

Application closes and stops all child processes.

# Observed behaviour

Several zombie child processes remain after application closes.

Here is full console output log showing processes remaining after closing MicroDrop app:

```sh
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3\microdrop-builder\Microdrop-win32-ia32> .\Microdrop.exe
... wait for app to open ...
... close app ...
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3\microdrop-builder\Microdrop-win32-ia32> Get-Process -Name Microdrop

Handles  NPM(K)    PM(K)      WS(K)     CPU(s)     Id  SI ProcessName
-------  ------    -----      -----     ------     --  -- -----------
    376      35    57444      62076       1.30   8260   1 Microdrop
    491     110    87344      94988       3.16   8696   1 Microdrop
    376      39    80436      93692       2.33  10264   1 Microdrop
    990      47    27640      61016     118.36  10268   1 Microdrop
    595      31    64692      66948     271.80  13196   1 Microdrop
    374      35    56688      62908       1.30  13716   1 Microdrop
```

## Note

Until a permanent fix is found, the zombie processes can be closed using the following command:

```sh
Get-Process -Name Microdrop | Stop-Process
```

Lucaszw, 2018-02-08:
Fixed

closed 2018-02-08 by Lucaszw

cfobel, 2018-02-08:
Fixed by cdfa07372d1a6b5cb95b5017c19bafaa40d5ba34