sci-bots/microdrop-3 #15: chore(electron-packager): use asar to combine node_modules into archive

cfobel, 2018-02-08:
As of 1e2e52281250ed3e135322d669237c9378458566, the [`microdrop-builder/electron-packager.config.js`][1] configuration does *not* use `asar` to write `resources/app/node_modules` to an archive, but rather, stores the `node_modules` in their original form.

This slows down install/uninstall operations considerably due to the many thousands of files in the `node_modules` file hierarchy.  In addition, it seems like the deeply nested files in the `node_modules` hierarchy may cause an issue for creating/installing a Conda package containing a MicroDrop Electron executable build.

--------------------------------------------

# Proposed solution

Add `asar` configuration to [`microdrop-builder/electron-packager.config.js`][1] to add `node_modules` to an `asar` archive in the built Electron app.

## Notes

 - We need to verify that static files (e.g., UI plugin `html` files) are still loaded as expected from the `asar` archive.  According to [here](https://stackoverflow.com/a/37603416/345236), it *might* "just work".

[1]: https://github.com/sci-bots/microdrop-3/blob/1e2e52281250ed3e135322d669237c9378458566/microdrop-builder/electron-packager.config.js

cfobel, 2018-02-09:
# Waiting on:

 - #20 chore(display): dynamically serve `display.html`
 - #21 chore(plugins.json): write dynamic plugins configuration to IndexDB

Lucaszw, 2018-02-09:
Update: Both #20 and #21 are closed now.

closed 2018-02-14 by Lucaszw