sci-bots/microdrop-3 #19: bug(application): Uncaught TypeError: Cannot match against 'undefined' or 'null'.

cfobel, 2018-02-09:
As of c8d95beab7d5f2566231f4884e3a5cf0a9d57366, the Electron app fails to launch with the following error:

[image]

Console log is:

```sh
(md-3-dev) PS C:\Users\chris\Miniconda2-x86\conda-bld\microdrop-3_1518178803860\work\microdrop-builder\packager\Microdrop-win32-ia32> .\Microdrop.exe
(md-3-dev) PS C:\Users\chris\Miniconda2-x86\conda-bld\microdrop-3_1518178803860\work\microdrop-builder\packager\Microdrop-win32-ia32>
Uncaught TypeError: Cannot match against 'undefined' or 'null'.
Uncaught TypeError: Cannot match against 'undefined' or 'null'.
Uncaught TypeError: Cannot match against 'undefined' or 'null'.
Uncaught TypeError: Cannot read property 'mqtt_ws_port' of null
```

cfobel, 2018-02-09:
Maybe this has to do with launching through `electron .\node_modules\@microdrop\application\bin\microdrop` instead of `electron .\node_modules\@microdrop\application\index.js` ?

Is there a reason why the code in `.\node_modules\@microdrop\application\bin\microdrop` cannot just be moved into `.\node_modules\@microdrop\application\index.js`?

Lucaszw, 2018-02-09:
This is fixed as of b5b67572133e72553df8e6c5b8de2b2a05b801f4. Closing

closed 2018-02-09 by Lucaszw