sci-bots/microdrop-3 #23: bug(electron-packager): `--help`  flag doesn't work as expected on Electron executable

cfobel, 2018-02-11:
## Run from source

When run from source, the `--help` flag gives the following output and stops execution (i.e., app does not continue to load):

```sh
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3\microdrop-builder> electron .\node_modules\@microdrop\application\bin\microdrop --help

{ env:
   { HTTP_PORT: 3000,
     MQTT_TCP_PORT: 1884,
     MQTT_WS_PORT: 8083,
     host: 'localhost',
     defaultEnabled: [ '@microdrop/device-ui-plugin', '@microdrop/state-saver' ] } }
usage: microdrop [-h] [-v] [-r] [--http-port HTTP_PORT]
                 [--mqtt-ws-port MQTT_WS_PORT] [--mqtt-tcp-port MQTT_TCP_PORT]

microdrop

Optional arguments:
  -h, --help            Show this help message and exit.
  -v, --version         Show program's version number and exit.
  -r, --reset           Reset application data
  --http-port HTTP_PORT
                        HTTP port (default 3000)
  --mqtt-ws-port MQTT_WS_PORT
                        MQTT websocket port (default 8083)
  --mqtt-tcp-port MQTT_TCP_PORT
                        MQTT TCP port (default 1884)
```

## Run from executable

When run from the packaged Electron executable, the `--help` flag gives the following output and continues to load the app:

```sh
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3\microdrop-builder\packager\Microdrop-win32-ia32> .\Microdrop.exe --help
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3\microdrop-builder\packager\Microdrop-win32-ia32>
{ env:
   { HTTP_PORT: 3000,
     MQTT_TCP_PORT: 1884,
     MQTT_WS_PORT: 8083,
     host: 'localhost',
     defaultEnabled: [ '@microdrop/device-ui-plugin', '@microdrop/state-saver' ] } }
{ args:
   Namespace {
     reset: null,
     http_port: 3000,
     mqtt_ws_port: 8083,
     mqtt_tcp_port: 1884 } }
Initializing Electrodes Model
Initializing Routes Model
Initializing Device Model
```

cfobel, 2018-02-12:
As of df7e91ba8e89058d7233080ea16a71e6eb4e2497, running the executable with the `--help` flag gives the following output and continues to load the app:

```sh
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3\microdrop-builder\packager\Microdrop-win32-ia32> .\Microdrop.exe --help
(md-3-dev) PS C:\Users\chris\Documents\GitHub\microdrop-3\microdrop-builder\packager\Microdrop-win32-ia32>
Initializing Electrodes Model
Initializing Routes Model
Initializing Device Model
```

cfobel, 2018-02-12:
@Lucaszw same behaviour as [above](https://github.com/sci-bots/microdrop-3/issues/23#issuecomment-364991757) as of 8fac092ca782bc56bdba8e26f2268820fc942b94.

Lucaszw, 2018-02-14:
Fixed with 227e75c4c811e8e88569cb7da5fd40e2a728851e

closed 2018-02-14 by Lucaszw