sci-bots/microdrop-3 #29: feat(plugins): (optionally) launch process plugins in Conda environment

cfobel, 2018-02-21:
Process plugins, e.g., Python, R, etc., may have other run-time requirements.

# Proposed solution

For processes where run-time requirements may be met through Conda packages, we should support executing the plugin in a Conda environment configured according to an **environment `.yaml` file**, which can either be [exported from an existing environment][env-export] or [created manually][manual].

## Implementation notes

Will at least require:

 - [ ] **feat(app)**: detection Conda installation (if available)
     * This can be done, e.g., by checking the `CONDA_PREFIX` variable
     * Maybe publish to MQTT topic?
 - [ ] **feat(plugin-manager)**: convention for process plugin to specify environment file, e.g., specify file name in plugin JSON config file.
 - [ ] **feat(plugin-manager)**: when launching process plugin:
     * [ ] If environment file is specified, create environment if it doesn't exist (and maybe apply [`conda env update ...`][env-update] if it does?)
     * [ ] Launch plugin process in activated plugin Conda environment

[env-export]: https://conda.io/docs/commands/env/conda-env-export.html
[manual]: https://conda.io/docs/user-guide/tasks/manage-environments.html#creating-an-environment-file-manually
[env-update]: https://conda.io/docs/commands/env/conda-env-update.html

cfobel, 2018-02-21:
@Lucaszw I updated the **"Implementation notes"** section in the description above:

>  * **Note: separate environments should probably be created for each `(<plugin_name>, <version>)`, e.g., with environment name `dropbot-1.3.2`, `dropbot-1.4.0`, since dependencies are likely to change between plugin versions**

Lucaszw, 2018-02-21:
I feel conda environments are great stream lined solution maintaining process plugins (due to the modular nature of environments). Here is a sample method I came up for how they could be handled (I will push this up with the Dropbot plugin shortly).

package.json:
```json
{
  "name": "@microdrop/dropbot-plugin",
  "scripts": {
    "install": "conda env create --force -f=./environment.yml -p ./conda",
    "start": "source activate ./conda && python dropbot_plugin/__init__.py",
    "dev": "yarn dev-p1 && yarn dev-p2",
    "dev-p1": "source activate ./conda && pip uninstall micropede || true",
    "dev-p2": "conda develop ../../../Micropede/packages/python -p ./conda"
  },
  "help": {
    "install": "create conda environment to hold python packages",
    "start": "run python script from withing conda environment",
    "dev": "link micropede package locally"
  },
  "publishConfig": {
    "access": "public"
  }
}

```

microdrop.json:
```json
{
  "name": "dropbot",
  "version": "0.0.1",
  "type": "process",
  "script": "yarn start"
}

```

[image]

cfobel, 2018-02-21:
What is the `-p ./conda` flag pointing to?

Lucaszw, 2018-02-21:
That is the path to the created conda environment (npm scripts always execute with the package.json file location as the cwd)

cfobel, 2018-02-21:
Cool, thanks!  I've never used that before, but that is really good to know.

So MicroDrop would just remain ignorant to Conda then?  i.e., `install` would just give an error if `conda` was not available within the environment MicroDrop is running in?

Lucaszw, 2018-02-21:
Rudimentary support adding in: d3c47b6c358db7b77161c8d4a4c3f6af7c301881

Here is the currrent workflow: 

During postinstall of microdrop/application, miniconda is installed silently.

If plugins want to use Microdrop's conda dist then they can specify like so in their package.json:

```json

{
  "name": "@microdrop/dropbot-plugin",
  "scripts": {
    "install": "$MICRODROP/miniconda/bin/conda env create --force -f=./environment.yml -p ./environment",
    "start": "source $MICRODROP/miniconda/bin/activate ./environment && python dropbot_plugin/__init__.py",
    "dev": "yarn dev-p1 && yarn dev-p2",
    "dev-p1": "source $MICRODROP/miniconda/bin/activate ./environment && pip uninstall micropede || true",
    "dev-p2": "$MICRODROP/miniconda/bin/conda develop ../../../Micropede/packages/python -p ./environment"
  },
  "help": {
    "install": "create conda environment to hold python packages",
    "start": "run python script from withing conda environment",
    "dev": "link micropede package locally"
  },
  "publishConfig": {
    "access": "public"
  }
}
```

The $MICRODROP variable is defined when the plugin is spawned (in @microdrop/application's webserver.js)

```javascript
          const env = process.env;
          env.MICRODROP = path.resolve(__dirname, '..');

          const options = {
            shell: true ,
            stdio: 'inherit',
            cwd: plugin.path,
            env: env
          };

          const runningChild = spawn(plugin.data.script, [], options);
```

Lucaszw, 2018-02-21:
It might be nice to the conda management into it's own repository (Yac : "Yet another conda package manager" / Yarn + Conda)

API could be somthing lile

``` bash
yac miniconda "miniconda name" # silently install new miniconda in cwd or given path
# ex. yac miniconda microdrop
yac locate "miniconda name" # locate a miniconda instance based on name
# ex. yac locate microdrop
yac init # create package.json with package name, and some field to signify this is a yac project
ex {
"name": "@microdrop/dropbot-plugin",
"yac": "microdrop",
...
}
yac install # creates local environment (yac_env) and local environment.yml
yac add "package name" # runs equiv of source activate ./yac_env && conda install pkgname  and adds to environment.yml
yac exec # executes scripts proceded by source activate ./yac_env
```

Lucaszw, 2018-02-22:
I updated the dropbot_plugin to use yac (https://github.com/sci-bots/yac).

closed 2018-02-22 by Lucaszw