sci-bots/microdrop-3 #82: style(ui): naming, menu reorganization

ryanfobel, 2018-04-12:
Some suggestions for naming, etc. to make things more clear:
* [ ] Change "Clear Routes" and "Execute Routes" context menu items to "Clear All Routes" and "Execute All Routes"
* [ ] Remove "Change Device" from the right-click context menu (this is now available in the main menu)
* [ ] Rename "Show" menu to "View" (I think this is a more common convention)
* [ ] Move "Open Dev Tools" from the Help Menu to the "View" menu and rename to "Dev tools"
* [ ] Change "transition-duration-milliseconds" to ""transition-duration-seconds"

Lucaszw, 2018-04-12:
" Remove "Change Device" from the right-click context menu (this is now available in the main menu)" 
: The menu is only available when running through electron, so removing it from the context-menu wouldn't allow for changing devices if we choose to eventually serve up the ui as a URL for raspberry pi, or via a jupter notebook / lab widget.

Lucaszw, 2018-04-12:
31ba38fd1c85f4f37f988b97084d28e63a9a9969

closed 2018-04-12 by Lucaszw